ttrftech/NanoVNA #6: License?

frohro, 2019-08-15:
Hi,
I don't find a license file here, or on your CentSDR page.  Is this an oversight?  
Thanks for your good work!
73,
Rob
KL7NA

frohro, 2019-08-15:
Hi,
I did find the GPL in the file source code files for this project.  It would be nice to have a LICENSE file in the repo though.  There is still nothing in the CentSDR that I could find either.  Maybe I should put that in the issues there, not here.  Sorry for that.
Thanks again,
Rob

RavinMaddHatter, 2020-07-07:
FYI, Not having a license means that technically nobody has permissions to use the source for anything. The license is what actually explains how we can use the product. By not putting a license it is actually as exclusive as possible, we have to assume that you want nobody to use any of the code and even copying small pieces of code can cause major legal issues.

If you put any license we can know what we are allowed to do., If you want people to do mostly what they want but still credit you i like the MIT license. There are for more restrictive licenses, but right now i cant even look at the project code without opening my self up to liability.

dl9sec, 2020-07-07:
Not any license, but the GPL, because parts of the code are already licensed under GPL-3.0.
To be most compliant, scan the sourcecode, extract all copyrights and all licenses, check license compatibility and put your code under the correct license. Put also all license texts that require it (most of the permissive licenses) into a license file and put this in the root directory of your repo.
If yo need assistance drop me a line :-)

Regards, Thorsten

dl9sec, 2020-07-09:
[Legal.zip](https://github.com/ttrftech/NanoVNA/files/4894633/Legal.zip)
If you want, you can put the files (all ASCII texts) into the root directory of the repository.
With those files (for v0.8.0), the project should be a bit more license compliant.

Regards, Thorsten