ttrftech/NanoVNA #22: USB Serial get corrupted data

cho45, 2019-08-31:
USB serial 経由のデータ(など)が化けることがあります。

- https://github.com/edy555/ChibiOS/blob/fe0ba1049c38346ceb2a396fa560848ef8323dd1/os/hal/lib/streams/chprintf.c#L121 tmpbuf がグローバルになっている
- shellThread の USB 経由で chvprintf が呼ばれる
- sweep スレッド (Thread1) の draw_all_cells() 中で chvprintf が呼ばれる (ここで tmpbuf が上書きされてしまう)

が原因かと思っています (実際 tmpbuf を stack 変数にすると再現しない)。が、tmpbuf は意図的に global にされているようなので、issue にします。

逆のケース (sweep スレッドの chvprintf が shellThread で上書きされる) もあると思いますが扱う文字数が違うので観測するのが難しそうです。

edy555, 2019-09-02:
メモリが溢れてた時、スタックの使用量を減らそうとグローバルにしてました。もはや不要なので戻しました。
https://github.com/edy555/ChibiOS/commit/669d4bbc8da1ee0e4ccdf93a472b06d183922320
当時はたまに壊れるのは気にしてなかったのですが、ツールとして常用すると気になりますね。

closed 2019-09-02 by edy555

cho45, 2019-09-03:
なるほど！ ご対応ありがとうございました。