ttrftech/NanoVNA #35: Crashing in version 1

mrobertsn, 2019-09-16:
I updated last night from the original stock firmware that came with the device. It was successful however every now and again the traces crash, as well as the screen, becomes unresponsive for a few seconds. Also sometimes parts of the traces persist on the screen for a few seconds while other parts are redrawn etc.
Also before the update, I was able to use the multi-segment sweep with VNAsaver pc app, but now it seems to always crash the device too.
When I say crash, the system just locks up and freezes with the screen freezing.

edy555, 2019-09-17:
Thanks for your report.
This problem is fixed by commit d68ab20c309ecedd625b658144f8b997a95285b0 and
you can try artifact in https://circleci.com/gh/ttrftech/NanoVNA/75#artifacts/containers/0

mrobertsn, 2019-09-17:
Would it also fix the crashes not caused by nanovnasaver? It also seemed a bit more laggy when inputting frequency ranges etc, any idea on the cpu utilisation of that little 48MHz CPU?

edy555, 2019-09-17:
The problem might be caused by the lack of stack area, I think.

closed 2019-09-20 by edy555