ttrftech/NanoVNA #45: Broken thru calibration

EB4FBZ, 2019-09-24:
Release 0.1.1 (commit d68ab20)

Bug found after updating my unit to this release. It worked fine with the stock firmware (GEN111.TAOBAO.COM). Double checked by flashing a backup.

After performing a full calibration (OPEN,SHORT,LOAD,ISO,THRU), the reference S21 shows a 0.5 to 1dB ripple. Ripple period depends on cable electrical length, but amplitude does not (tested with different quality cables).

 If only ISO+THRU or just THRU calibration is done, the trace is flat.

EB4FBZ, 2019-09-24:
Same ripple present in 9508ea7. Attached photo.
[image]

EB4FBZ, 2019-09-25:
More examples using both firmwares, allways using the original OSL kit but a quality RG316 30cm pigtail and quality adapters:

- First test: calibrating for reflection at the end of the cable, using a F-F SMA adapter. Removing the adapter for THRU calibration.

  - Original firmware: [image]

  - Commit 9508ea7: [image]

- Second test: calibrating for reflection at the end of the cable, using a F-F SMA adapter. But this time i add a male-male adapter for THRU calibration.

  - Original firmware: [image]

  - Commit 9508ea7: [image]

- Third test: calibrating for reflection right at the CH0 SMA. Adding the pigtail for THRU calibration.

  - Original firmware: [image]

  - Commit 9508ea7: [image]

It's clear to me that something weird is happening in newer firmwares. Some wrong interaction between reflection and transmission cal coeffs maybe? If i reset calibration and run a THRU cal without OSL steps, S21 response is perfectly flat.

qrp73, 2019-09-26:
Cannot reproduce with the latest change set. 
Hardware: NanoVNA-H
CH0 to CH1 cable: 10 cm RG405, soldered SMA

I have 0.2 dB error at 0.05 MHz after calibration, but it is almost flat and smoothly going to 0 dB error at 900 MHz.

qrp73, 2019-09-26:
When I use RESET and then CALIBRATE THRU only I get almost zero S21 error from 0.05 MHz to 900 MHz.

qrp73, 2019-09-26:
I tried to split calibration into two stages:
1) OPEN, SHORT, LOAD, DONE
2) ISOLN, THRU, DONE

And got S21 error almost zero from 0.05 MHz to 900 MHz. So, it seems like temporary workaround for a proper calibration.

EB4FBZ, 2019-09-26:
I think you are experiencing the same error, but with a shorter cable the ripple period is much longer and not so evident. 

You can't calibrate in two stages, starting a new calibration resets previous one.

EB4FBZ, 2019-09-26:
More examples, this time including phase trace:

- Stock firmware with full calibration:
[image]

- Commit 9508ea7 with full calibration. OSL al the end of the cable and then add a male-male adapter for THRU:
[image]

- Commit 9508ea7 with full calibration but reflection calibration just at the CH0 connector. Then add cable for THRU:
[image]

- Commit 9508ea7 with THRU calibration only:
[image]

Something is wrong.

cho45, 2019-09-26:
This may caused by difference between NanoVNA and NanoVNA-H.

NanoVNA-H does not use ETERM_ES (source match) for calculating ETERM_ET (transmission).

NanoVNA: https://github.com/ttrftech/NanoVNA/blob/master/main.c#L1058
```
// CAUTION: Et is inversed for efficiency
static void
eterm_calc_et(void)
{
  int i;
  for (i = 0; i < sweep_points; i++) {
    // Et = 1/(S21mt - Ex)(1 - Es)
    float esr = 1 - cal_data[ETERM_ES][i][0];
    float esi = -cal_data[ETERM_ES][i][1];
    float s21mr = cal_data[CAL_THRU][i][0] - cal_data[CAL_ISOLN][i][0];
    float s21mi = cal_data[CAL_THRU][i][1] - cal_data[CAL_ISOLN][i][1];
    float etr = esr * s21mr - esi * s21mi;
    float eti = esr * s21mi + esi * s21mr;
    float sq = etr*etr + eti*eti;
    float invr = etr / sq;
    float invi = -eti / sq;
    cal_data[ETERM_ET][i][0] = invr;
    cal_data[ETERM_ET][i][1] = invi;
  }
  cal_status &= ~CALSTAT_THRU;
  cal_status |= CALSTAT_ET;
}
```

NanoVNA-H: https://github.com/hugen79/NanoVNA-H/blob/master/main.c#L1103 
```
// CAUTION: Et is inversed for efficiency
static void
eterm_calc_et(void)
{
  int i;
  for (i = 0; i < sweep_points; i++) {
    // Et = 1/(S21mt - Ex)(1 - Es)
   // float esr = 1 - cal_data[ETERM_ES][i][0];
    //float esi = -cal_data[ETERM_ES][i][1];
    float s21mr = cal_data[CAL_THRU][i][0] - cal_data[CAL_ISOLN][i][0];
    float s21mi = cal_data[CAL_THRU][i][1] - cal_data[CAL_ISOLN][i][1];
//    float etr = esr * s21mr - esi * s21mi;
//    float eti = esr * s21mi + esi * s21mr;
//    float sq = etr*etr + eti*eti;
//    float invr = etr / sq;
//    float invi = -eti / sq;
    float sq = s21mr*s21mr + s21mi*s21mi;
    float invr = s21mr / sq;
    float invi = -s21mi / sq;
    cal_data[ETERM_ET][i][0] = invr;
    cal_data[ETERM_ET][i][1] = invi;
  }
  cal_status &= ~CALSTAT_THRU;
  cal_status |= CALSTAT_ET;
}
```

qrp73, 2019-09-26:
cho45, I also notice these changes in calibration calc, but I'm not sure - which version is better?

edy555, 2019-09-26:
I found a failure. The expression in comment L1074 `// Et = 1/(S21mt - Ex)(1 - Es)` is incorrect, missing multiplier S11 for Es. The right is `Et = 1/(S21mt - Ex)(1 - Es S11)`.
And assume S11 is 1 in treating thru as ideal, it becomes `Et = 1/(S11mt - Ex)`.

In short, the fixed code in NanoVNA-H is right.

edy555, 2019-09-26:
Just fixed at https://github.com/ttrftech/NanoVNA/commit/0b885f264078d11ab3f95ef74c9b6cd6aa146956

Please try recent master branch or binary in https://app.circleci.com/jobs/github/ttrftech/NanoVNA/84/artifacts

EB4FBZ, 2019-09-27:
It's fixed! Thanks a lot @edy555

qrp73, 2019-09-28:
yes, I can confirm, now error is smaller than 0.1 dB across entire bandwidth 0.05 - 900 MHz. For 0.05-300 even better - smaller than 0.01 dB :)

Thanks