ttrftech/NanoVNA #46: averaging and i/f bandwidth selection support

edy555, 2019-09-24:


edy555, 2020-01-14:
作ってみたけど、測定結果に違いは見えない。
https://github.com/ttrftech/NanoVNA/tree/bandwidth

[image]

d51r3verse, 2020-03-31:
100Hz traces more linear/quieter than 1kHz( bandwidth seems to work I think 😁  )
But, unexpected readings found `bandwidth`ed calibration&sweeps.
Here is my experience( both ends terminated LOAD )

Cal:1kHz / Sweep:1kHz
[image]

Cal:100Hz / Sweep:100Hz
[image]

Pmax65, 2020-03-31:
Hi to all, 

what is the sweep time for the 100Hz RBW?

I don't want to "teach the cat climbing to the roof", but remember that a 100Hz filtering have a longer dwell  time, if you don't reduce the sweep time according to the filter dwell time you may get just wrong measurements.

Have a great day.

Massimo