ttrftech/NanoVNA #47: F303 support

edy555, 2019-09-24:
To easily extend features, F072 is not enough to extend something valued.

AA6KL, 2019-09-26:
I am trying to port your code to F303.  My tree is at

https://github.com/AA6KL/NanoVNA

I am not an expert in embedded software, nor experienced in ChibiOS.  Please take a look at my code and merge to yours if you think my change is correct.  

Ken
AA6KL

edy555, 2019-09-28:
Many thank for your challenge to this issue and sharing.

I used F303 in [another project](https://github.com/ttrftech/CentSDR), so I'll use my own. 
In that project, initializing si5351 in the bootstrap sequence to obtain MCLK, so 8MHz external crystal is not necessary.

Please keep trying your challenge. Please let me say that thank you again.