ttrftech/NanoVNA #54: frequency error caused from fractional pll

edy555, 2019-09-28:
Frequency error exists between LO and REF, which is around some numbers of 10Hz. It caused by fractional PLL which use inappropriate divisor.

qrp73, 2019-09-29:
Hi, edy555, there is related question. I'm not sure why there is limit for the minimum frequency span? With the latest firmware the minimum span is 10 kHz, when I'm trying to set 1-9 kHz it shows saw wave. 10 kHz span is insufficient for some tasks, such as crystal measurements, so it will be great to reduce this limit to allow at least 1 kHz span.

si5351 allows to produce frequency with very high resolution, much better than 1 Hz. it is used in another VNA projects. If there is needs help, I can share tested code for si5351 which works great.

Is it possible to change this frequency span limitation to at least 1 kHz span?
Thanks

edy555, 2019-09-29:
The issue about narrow span is addressed in #52 but fix in 0.2.0 release is incomplete, as you said.
I just fixed it again in https://github.com/ttrftech/NanoVNA/commit/0aeadaa213e16c60e3044fd4cccb45fe60551714
please try https://github.com/ttrftech/NanoVNA/releases/tag/0.2.2

qrp73, 2019-09-29:
yes, now narrows span works, thanks :)

edy555, 2019-12-09:
fixed at https://github.com/ttrftech/NanoVNA/commit/6a88f8ed8f18e09f1976db74359a76049c185470

closed 2019-12-09 by edy555