ttrftech/NanoVNA #55: 0.2.1 release hangs on trace selection

N3GO, 2019-09-28:
Downloaded and installed (2 attempts) firmware hangs on menu selection of any Trace.

edy555, 2019-09-28:
Uhmm, I cannot reproduce the failure.
Please let me know the detail of the procedure.

N3GO, 2019-09-29:
I think this will do it... Blank all of the traces on the screen, and 
save that state to Store 0. Reset the NanoVNA, then try to turn one or 
more of the traces.

qrp73, 2019-09-29:
Cannot reproduce. Tested with the latest commits. Works as expected.

I tried the following STR (steps to reproduce):
1) Open DISPLAY=>TRACE menu and disable all traces
2) Open CAL=>SAVE menu and click on SAVE 0
3) Power off and the power on
4) Open DISPLAY=>TRACE menu and enable all traces
5) Make sure all traces again visible and works ok

N3GO, 2019-09-30:
I replicated the procedure as posted to verify the sequence induces the failure...

It fails at Step 4.

It highlights Trace 1 when I select it, then hangs and fails to respond to any further input. It will not allow me to go back in the menu, and touching elsewhere on the screen doesn't close the trace menu list.

I downloaded a fresh copy of nanovna-firmware-0.2.2.zip, created and installed a new DFU file and got the same result. 

I created the DFU file from the hex code (ch.hex) in the build folder using DFU File Manager (v3.0.6), and ported the DFU result to the NanoVNA via DfuSe Demo (v3.0.6) all on a Windows 10 desktop. This procedure fails only with this version, so I'm reluctant to "blame" this on any issues with DFU File Manager and/or DfuSe Demo. 

A reinstall is also insufficient to recover. I must return to a previous version (e.g. v 0.1.1), cycle power, and then reinstall v.0.2.2.

Could there be some residual data from previously tested versions that doesn't get cleared before updating with new firmware?   

72

Gary, N3GO

EB4FBZ, 2019-10-05:
Reproduced in 0.2.2. Trying to enable a disabled trace (saved as disabled in 0) hangs the unit.

edy555, 2019-10-05:
I tried again but cannot reproduce.

EB4FBZ, 2019-10-05:
A friend reproduced it on his unit.

- Disable Trace 3
- Save
- Shut down the VNA
- Power up the VNA
- Try to enable Trace 3. The unit will hang.

N3GO, 2019-10-10:
Closing this Issue. It appears to have been fixed in 0.2.3 release.

closed 2019-10-10 by N3GO