ttrftech/NanoVNA #57: touch freze

qrp73, 2019-09-29:
STR:
1) power on device
2) during device initialization, touch at the center of screen with frequency 2-3 times per second
3) make sure device freezes and don't respond on any input anymore.

This bug is a little floating and may require 2-3 times to repeat in order to reproduce it.

edy555, 2019-09-29:
I could have reproduced the issue. I'll investigate it.

qrp73, 2019-09-29:
it happens in the plot.c, line 941 (function search_index_range_x, first while loop).

It seems that this loop is infinite, because the value is not found and there is no way to exit from the loop in such case...

edy555, 2019-09-29:
bingo! Fixed in https://github.com/ttrftech/NanoVNA/commit/30d33571fa3929ff697bf410d3b7f25145cc6e45
Thank you!!

qrp73, 2019-09-29:
confirmed. Thanks

closed 2019-09-29 by qrp73