ttrftech/NanoVNA #59: strange "peaks" in S11 after upgrading to 0.2.1 and higher

df6dbf, 2019-09-29:
Have here "strange" Peaks after upgrading to v0.2.1. v0.2.2 shows same behaviour. Took Screenshots with nanoVNA-saver to show the Problem. The v0.2.0 from 27. Sept. seems ok.
[image]
[image]

EB4FBZ, 2019-09-29:
I can't reproduce this issue. Just updated from 0.1.1 to 0.2.2 and the response of a filter is the same in both versions.

qrp73, 2019-10-01:
cannot reproduce

gaionim, 2019-10-05:
> Have here "strange" Peaks after upgrading to v0.2.1. v0.2.2 shows same behaviour. Took Screenshots with nanoVNA-saver to show the Problem. The v0.2.0 from 27. Sept. seems ok.

are You using multiple segments in nanovna-saver ?
I have similar Peaks, but only in nanovna-saver with multiple segments.

df6dbf, 2019-10-05:
Yes, its occured only in Multi-Segment Mode.

edy555, 2019-10-07:
This issue is caused by behavior change of `sweep` command at 0.2.0.

This commit https://github.com/ttrftech/NanoVNA/commit/3fc17e18f900c6bb31fb1e72bab345adf576c15c shall make equivalent to former behavior. When I tested NanoVNA-saver 0.1.0 in multisegment mode with new firmware, it works well.

New [release 0.2.3](https://github.com/ttrftech/NanoVNA/releases/tag/0.2.3) is available including commit above.
Could you check it out?

df6dbf, 2019-10-07:
Installed the 0.2.3 and tested with actual Version of nanoVNA-saver. It seems the problem is solved. Thank you! :)

closed 2019-10-07 by edy555