ttrftech/NanoVNA #60: Make max  SI5351 output frequency user settable

erikkaashoek, 2019-09-30:
There is some variation in the quality of the SI5351 being used. Going to 300MHz is formally always out of spec so you are never sure if the nanoVNA will be able to reach 300MHz
Would it be an idea to make the max SI5351 output frequency user settable (and stored in flash) so users experiencing stability issues can set the max SI5351 frequency to a lower value?

edy555, 2019-09-30:
Thanks for pointing out the issue. I also have realized it. https://twitter.com/edy555/status/1173029790808301569
I'll implement a configuration command to adjust the threshold of frequency using harmonic mode.

edy555, 2019-10-05:
Added `threshold` command to configure maximum output frequency in commit https://github.com/ttrftech/NanoVNA/commit/94659a22ba90955a8eb82657bceaff6daa7a96c6
Needs `saveconfig` to store into flash after `threshold` command.
Please check it out!

erikkaashoek, 2019-10-06:
Works very well!
Will post updated build on the nanoVNA group.io

closed 2019-10-06 by erikkaashoek