ttrftech/NanoVNA #64: missing group delay and phase delay measurements

qrp73, 2019-10-05:
There is exists trace format DELAY in the menu, but it doesn't works, because it is not implemented in the code.

I implemented group delay (GD) in this changeset: https://github.com/ttrftech/NanoVNA/pull/48/commits/e81f4fe82872c434b58e2cb87263d79a07322156

Also it will be nice to add phase delay (PD) trace. But it requires continuous phase unwrap for entire sweep, so I'm not sure how to implement it with no use additional memory.

edy555, 2019-10-05:
Thanks for creating separate PRs.

Currently, format group delay is implemented in branch https://github.com/ttrftech/NanoVNA/commits/format_delay
and it will be merged.
But seeing your implementation I feel interesting and helpful.
Thank you!

qrp73, 2019-10-11:
Thanks, it will be nice to add phase delay (PD) trace

closed 2019-10-11 by qrp73