ttrftech/NanoVNA #69: floating point data loss due to chprintf

qrp73, 2019-10-06:
chprintf doesn't supports lossless floating point formats such as "%.9g". It seems that it has very basic support for floating points. As result, the floating point value is truncated with data loss, especially for very small floating point numbers.

I propose to use snprintf instead of chprintf and then put the buffer to the output stream. 

For example:
```
        char tmpbuf[32];
        int leng = snprintf(tmpbuf, sizeof(tmpbuf), "%.9g", s11[0]);
        for (int j=0; j < leng; j++) {
            streamPut(chp, (uint8_t)tmpbuf[j]); 
        }
        streamPut(chp, (uint8_t)'\t'); 
        leng = snprintf(tmpbuf, sizeof(tmpbuf), "%.9g", s11[1]);
        for (int j=0; j < leng; j++) {
            streamPut(chp, (uint8_t)tmpbuf[j]); 
        }
        streamPut(chp, (uint8_t)'\t'); 
```

edy555, 2019-10-06:
Thanks for your comment, your point out must be true.
But `snprintf` needs to link `stdio` and heap, and current firmware doesn't have heap and the shortage of flash or SRAM doesn't allow to use them.

As I know, HP8753 ought to use hexadecimal to read/write floating-point data. This is an available choice.

qrp73, 2019-10-07:
I implemented custom scanraw command which perform unlimited amount of point count measurement and uses snprintf and it works great. It needs a little more deep stack size. The max stack usage is about 502 for my case, so I set 502+16=518 instead of 442.