ttrftech/NanoVNA #70: first measurement point sometimes has wrong value

qrp73, 2019-10-06:
There is strange bug. Sometimes the first measurement point has very high spike.

I found the following STR that allows to reproduce this spike with high probability:

1) power on
2) open CONFIG -> VERSION screen and wait 5-10 sec
3) tap on the center of screen to close version screen
4) note that first sweep has high spike at first measurement point

The same spike may happens when you use joystick to move cursor left or right

edy555, 2019-10-06:
Thanks for your report on your observations. I have also realized this issue but not investigated it yet.

I wonder whether does your patch https://github.com/ttrftech/NanoVNA/commit/23dba7d9b4dce23889ec25e39c922cc9973daf82 make it better or not. Does it have any effect on this issue?

qrp73, 2019-10-07:
the patch make measurement very stable at frequency boundaries (300 MHz, 600 MHz, 900 MHz), but this issue with first measurement point still exists. There is also interesting observation - for some reason this issue affects first measurement point only.

I think something going wrong at i2s_end_callback. May be race condition with wait_count, but I'm not sure. Needs to investigate it more deep.

qrp73, 2019-10-08:
fixed :)

edy555, 2019-10-08:
Modification that you pointed out is very helpful to fix. Thank you!