ttrftech/NanoVNA #79: Numpad editor freeze bug

qrp73, 2019-10-12:
STR:

1) Open any numpad editor, for example DISPLAY => SCALE => ELECTRONIC DELAY
2) Touch right bottom corner of screen
3) Make sure there is a rendering bug and numpdad freeze, there is no way to continue, the only way is to power off device.

There is also missing way to cancel numpad editor.

reald, 2019-10-12:
Workaround: You can move the wheel switch left or right to get the numpad back.

edy555, 2019-10-13:
I have noticed that the number pad or user interface for numeric input is not good and am thinking this and that to be better.
Anyway, thanks for your comment.

ssd2, 2019-10-17:
The touchscreen is a bit twitchy (understandable), but I've accidentally triggered the touchpad a few times and couldn't cancel it without changing the value.