ttrftech/NanoVNA #88: Use 1 16bit word to define firmware capabilities - ENHANCEMENT REQ

nlroth, 2019-10-23:
Is it possible for the 4-5 main developers of the various NanoVNA firmware flavours to agree on a simple identification word that would be included in the VERSION console command?
This single 16bit word would be able to identify the presence (or absence) of up to 32 items. There could be more items defined if sub-bits are used in a binary counting fashion. 
For example:
Bit 0    can be a 0 or 1 to define the presence of a feature
Bits 1, 2 & 3 would allow 8 sub-levels of the preceding feature.

The problem is not in defining the table, bit it is having the table implemented and followed by the various NanoVNA firmware developers.

Please refer to forum post https://groups.io/g/nanovna-users/message/5610  where I have attached an Excel sheet as an example.

This type of ID 'string' would be very helpful in the case of a Black Box version of the NanoVNA.

edy555, 2019-10-24:
Thanks for the proposal.
I think that using strings of expressing features is better than a number of 16bit integers.
I'd like to add a new command for that.

nlroth, 2019-10-24:
QRP suggested using both the help and version commands to determine features.The drawback is ensuring the help command string is kept up to date as other modules are updated.
Personally, I like to keep things simple (My 1977 SOL20 had 2K ram and 1K of that was the display - it is amazing what you can squeeze into 1K ram :-) )