ttrftech/NanoVNA #89: Characteristic values guidelings on SWR graph

pedjas, 2019-10-24:
It would be good to have guidelines for SWR 1:1.5 and 1:2 on SWR graph. Those are characteristic values for SWR. It would help a lot readability of SWR graph.

Guidelines could be just simple horizontal lines on graph.

edy555, 2019-10-24:
In default, SWR format graph has 1.0/div of scale value, 1:2 is on the div line.
If you select 0.1 or 0.5/div of scale, SWR 1:1.5 value comes onto the div line. It is enough to read the SWR graph.
I think that VNA is not an antenna analyzer, so the SWR graph should not be treated special.

closed 2019-10-24 by edy555

pedjas, 2019-10-24:
The same principle could be followed for other measurements. It is always need to set some specific value that is looked upon, and it would be nice to be able to see it easily.