ttrftech/NanoVNA #90: Calibration of voltage level indicator

pedjas, 2019-10-24:
It is reported that battery voltage measurement is not precise due to various didoes used to add this feature.

It would be good to have an option to calibrate level measurement by letting user measure actual battery voltage and then set that voltage as calibrated value. 

This does not even need graphical interface. Calibrating using console would be ok, as it is needed just once - after diode is installed.

edy555, 2019-10-24:
Thank you for proposal. 
However, I think, it is caused that customers use various diodes as D2. It should be a specific diode when manufactured.
Flash memory is very short, so there is no room for it. I will not add that feature.

closed 2019-10-24 by edy555

zstadler, 2019-11-02:
@edy555,

It is my understanding that the NanoVNA LED stops blinking when the battery is fully charged.
If the firmware is responsible for the blinking, would it be possible to sample the D2 voltage when the battery is fully charged and use that value instead of manual calibration?

If this approach is possible, it would have several advantages:
- Many NanoVNA owners have devices that were manufactured without a D2 and we would like to see a battery indicator.
- This approach avoids the need to find the specific calibration value followed by creating new firmware and flash it after installing a D2 diode.
- This approach is independent of the electrical characteristics of different D2s. 
- Re-calibration would be done whenever the battery is fully-charged and will be consistent with the LED blinking.

pedjas, 2019-11-02:
NanoVNA-Q 0.4.3 added vbat_offset command (allows to set battery diode offset)

qrp73, 2019-11-07:
> If the firmware is responsible for the blinking

firmware is not responsible for charge diode blinking. 
It is done inside charge controller IP5303 and don't depends on firmware at all.