ttrftech/NanoVNA #91: Vbat negative and config not restored

helgein, 2019-10-28:
Firmware: 0.3.1-20191021

**Config is not stored, battery icon is empty, vbat returns -1.**
Battery works fine and powers the device for multiple minutes without usb cable plugged in.
After power cycling, all display related configuration of traces, format, channels are lost 
and a "weird default" shows all 4 traces.

Flashed the "Eraser":
```
C:\Dell>"C:\Program Files (x86)\STMicroelectronics\Software\DfuSe v3.0.6\Bin\DfuSeCommand.exe" -c -d --fn FILLFF.dfu
...
```
Flashed the image:
```
C:\Dell>"C:\Program Files (x86)\STMicroelectronics\Software\DfuSe v3.0.6\Bin\DfuSeCommand.exe" -c -d --fn ch.dfu
1 Device(s) found :
Device [1]: STM Device in DFU Mode, having [2] alternate targets
Target 00: Upgrading - Download Phase (100)...Duration: 00:00:10
Upgrade successful !
```
* Power-cycled the device. Empty battery-icon shown.
* in the terminal, executed 
```
ch> clearconfig 1234
Config and all cal data cleared.
ch> saveconfig
Config saved.
ch> vbat
-1 mV
```
* Calibrated it, stored to "0".
* Configured to just show traces 0 and 2 as smith for ch0 and ch1.
* saved the config from the touchscreen as well as from the terminal via "saveconfig" command.
* vbat negative:
```
ch> vbat
-1 mV
```
tried terminal command "clearconfig" but it did not change a thing.

edy555, 2019-10-30:
- Did you install D2? It is needed to show the battery indicator.
- Try to cycle power after `clearconfig`, do not do `saveconfig`.

qrp73, 2019-11-06:
it seems that it's better to make reset after clearconfig, because some peoples trying to do saveconfig just after clearconfig, which leads to write wrong config back. I already seen such reports

edy555, 2019-11-07:
added guide text after doing clearconfig at https://github.com/ttrftech/NanoVNA/commit/e6035a5d96d0fea00d4cd7d5479a8ce865ebd38b

ZeroChaos-, 2020-03-09:
I have done a clearconfig and power cycle but vbat shows -1 and I have no battery icon at all.  I am using c40d78d80fcba820a9260cfe16b83349537f9a34 (today).

I bought a nanovna off amazon from some no name seller, so I can't really say much about the board other than the battery indicator used to work.

RolandWa, 2020-03-09:
Hi Zero,

You have to check if you have soldered diode to the power supply sampling
point, see the web page below :

https://www.google.com/amp/s/dw1zws.com/enabling-battery-status-indicator-on-nanovna/amp/

Roland

ZeroChaos-, 2020-03-09:
thanks for the pictures, that is my problem.