ttrftech/NanoVNA #93: Cross compiling on a Raspberry Pi...

dl9sec, 2019-11-03:
Hi,

i tried to cross compile the stuff on a Raspberry Pi 3. The toolchain mentioned at "Prepare ARM Cross Tools" can not be used on a RasPi.
But i found a precompiled toolchain for the Raspberry Pi at [https://github.com/vanbwodonk/gcc-arm-embedded-build-armhf](url). This toolchain is more recent (8-2019-q3-update).
Differing from the description there, i used the following lines to install the toolchain:

`wget https://github.com/vanbwodonk/gcc-arm-embedded-build-armhf/raw/master/pkg/gcc-arm-none-eabi-8-2019-q3-update-linux-armv7l.tar.bz2`
`tar -xvf gcc-arm-none-eabi-8-2019-q3-update-linux-armv7l.tar.bz2`
`sudo cp -r gcc-arm-none-eabi-8-2019-q3-update/* /usr/local/`
`sudo rm -r gcc-arm-none-eabi-8-2019-q3-update`

Additionally i added the following line at the end of ~/.profile:

`PATH=/usr/local/usr/local/arm-none-eabi/bin:$PATH`

My first attempt fails to get the toolchain work  on my Raspbian Stretch (Debian 9) because the toolchain needs glibc 2.28, but Stretch uses 2.24.
So Raspbian Stretch must be updated to Buster (see https://pimylifeup.com/upgrade-raspbian-stretch-to-raspbian-buster/) to get the toolchain work.

After upgrading Raspbian `make` runs well and the binary was built.

Hope this helps if someone want to cross compile on a RasPi...

Regards, Thorsten

JulyJim, 2019-11-17:
I am currently using local repository of NanoVNA  and Eclispe IDE. 
Combination of git and EGit (Eclipse plug-in) really does not yield to easy actual code debugging. 
I have successfully used TCF (Target Communication Framework ) on RPi on another project.
I'll give your GREAT idea try. THANKS

ehaag, 2019-11-21:
This is really great, thank you.

JulyJim, 2019-11-21:
Not really, I still cannot analyse / debug the software. 
I get two errors - if run locally - cannot execute .efl file
if run on RPi I get I/O error loop and cannot find from where...
I realize I cannot run STM32 I/O on RPi, but sure like to just set breakpoint on "main.c" and print the variables contents on standard console. Is that so much to ask?
There is "printf" debug option on IDE /TCF  I am trying got implement.