ttrftech/NanoVNA #94: separate display format from calibration set in saved settings

ssd2, 2019-11-04:
I understand why the sweep range is saved with the calibration settings, but it would be nice if the display format (enabled traces, trace formats, units/division, etc.) were saved separately.

That way, I could have display formats saved for each use mode; I.e., antenna measurement, TDU, etc.   And be able to recall calibration ranges without changing display format, or change display format without either changing calibration range or manually changing each trace...

It probably makes sense to keep the markers with the calibration and range.

qrp73, 2019-11-06:
that's bad idea, because settings are stored in the program flash, which has large page size and limited write cycle limit. If you split trace settings and calibration data, it will leads to twice more program memory requirements. Since there is already no space, you will needs to remove TDR and something other in order to implement such feature. Such change will be useless, because it's not hard to change trace settings, but remove TDR and some other functions is a major regression.

ssd2, 2019-11-06:
This wouldn't take any more flash memory than is in use now.
Is program memory so tight that another menu wouldn't fit?  I'd certainly not want to remove other features.

If that's the case, here's hoping the next version uses a cpu with more program memory!