ttrftech/NanoVNA #96: New firmware 0.4.0 doens't work as expected...

dl9sec, 2019-11-07:
Hi,

got my NanoVNA two days ago with a well working, but unknown version firmware:

[image]

I backed up 128kB of the flash with dfu-util and tried the latest v0.4.0. The firmware runs, but the measurement seems to be broken:

[image]

A calibration doesn't change anything.
If i restore the firmware with my backup, all runs well again, measurements are plausible.
So i assume, the hardware itself works.

Did i miss something to make the measurement work? Some kind of factory resetting or so?
Any ideas?

Thanks in advance,

Thorsten

edy555, 2019-11-08:
Could you try command `clearconfig 1234` after flashing?

insanumingenium, 2019-11-09:
I had the same exact scenario, was banging my head against garbage starting around 300Mhz regardless of calibration.  Below that it was working normally, and nanovnasaver seemed to give better results (didn't test extensively).

Clearing worked for me, now seems to be working 100%.

dl9sec, 2019-11-09:
Confirm. Tnx :-D

closed 2019-11-13 by edy555