ttrftech/NanoVNA #99: Smith value format is not persistent

owenduffy, 2019-12-10:
It appears that selecting Smith value format R+jX does not persist over power off.

edy555, 2019-12-13:
Yes, currently smith value format is not stored but will be supported in near releases.

owenduffy, 2019-12-13:
It could be argued that the default format should be R+jX.

Further, that your Ω symbol should be on the right hand end of the value, there should be no space between real and imaginary parts (with a + sign to separate as needed), and the J should be j (J is for joules) and be between the imaginary parts sign and its numeric characters.

Screen space is very valuable, so why not abbreviate SMITH to SM? If there is no facility to scale the Smith chart display, what is the value of the "1.0FS" display?

Adding this feature is a significant improvement on the hammy presentation, fixing these issues is as big an improvement.

Thanks for the good work.

Owen

jp3141, 2019-12-30:
so the format should look like:
`49.2-j2.87m Ω` // only space is between the last character of the value and the Ω symbol

owenduffy, 2019-12-30:
Quite, but screen space is a premium so I have no problems with no space between value and unit (as is done with other quantities).

closed 2020-01-16 by owenduffy

edy555, 2020-01-18:
Smith value format became persistent in 0.6.0 release https://github.com/ttrftech/NanoVNA/releases/tag/0.6.0