ttrftech/NanoVNA #101: What is limiting the frequency lower bound ?

ks156, 2019-12-21:
Hello,

I'm trying to figure out what is limiting the lower frequency to 50kHz.
I think that the limitation is coming from the hardware, is it correct ?

I'm wondering if there's a way to decrease this limitation around 100Hz ?

edy555, 2019-12-25:
The limitation is coming from architecture mainly. And also limited by coupling capacitor of mixer input by hardware.

The lower bound is limited by IF frequency. It needs twice or more of IF frequency.
But 50kHz is not a specific value, it is decided as appropriate for RF instruments. It must be work on 20kHz.

If you need an instrument work on audio frequency, architecture must be changed.
TX signal should be generated from an audio codec and RX signal should not be processed by mixer but converted by ADC directly.

ks156, 2019-12-30:
@edy555 thank you for this complete answer.
I was expecting (dreaming, maybe) that the lower frequency could be reduced easily. But it seems that it is more difficult than just changing one or two components and chaning two or three lines in the firmware 😄