ttrftech/NanoVNA #102: Missing Autoscale

damib, 2019-12-24:
Hi

I find this project really amazing, I usually use VNAs to test RF cavities, and one of the most useful feature I use to quickly find the correct scale/refpos, is autoscale.
I've already tested some code in order to add this feature, are you interested to consider it? I would like to have an opinion from the admin before posting a pull request...

edy555, 2019-12-25:
Interesting!
Please show me as PR.

damib, 2019-12-26:
Ok

damib, 2019-12-26:
I've just realized that it does not fit in the memory of the master branch. Since the beginning I did all the work starting from the merge #79 where you reduced the flash usage

current master:

   text	   data	    bss	    dec	    hex	filename
  92596	   5040	  11536	 109172	  1aa74	build/ch.elf

Merge 79:

   text	   data	    bss	    dec	    hex	filename
  88044	   5032	  11544	 104620	  198ac	build/ch.elf

Merge 79 + autoscale

   text	   data	    bss	    dec	    hex	filename
  89276	   5032	  11544	 105852	  19d7c	build/ch.elf

So we can postpone the PR after the merge 79 will be included in the master branch. In the meanwhile I try to reduce the autoscale footprint as much as I can.

If you want to look the code you can find it here https://github.com/damib/NanoVNA/tree/autoscale

damib, 2019-12-27:
With LTO merged in the master branch:
   text	   data	    bss	    dec	    hex	filename
  90376	   5032	  11544	 106952	  1a1c8	build/ch.elf