ttrftech/NanoVNA #109: LTO cause freeze

edy555, 2020-01-09:
Enabling the LTO option to reduce flash size causes freeze occasionally.
So, unfortunately, unable to merge the autoscale feature contributed from damib.

damib, 2020-01-10:
Ok, I'll keep on trying to find an alternative, thanks.

andyg24, 2020-01-20:
When linking with LTO enabled, gcc spits out the following warnings:

Linking build/ch.elf
ui.c:308:13: warning: type of 'ili9341_line' does not match original declaration [-Wlto-type-mismatch]
 extern void ili9341_line(int, int, int, int, int);
             ^
ili9341.c:403:1: note: type mismatch in parameter 5
 ili9341_line(int x0, int y0, int x1, int y1, uint16_t fg)
 ^
/usr/include/newlib/sys/_stdint.h:36:20: note: type 'uint16_t' should match type 'int'
 typedef __uint16_t uint16_t ;
                    ^
ili9341.c:403:1: note: 'ili9341_line' was previously declared here
 ili9341_line(int x0, int y0, int x1, int y1, uint16_t fg)
 ^
ili9341.c:403:1: note: code may be misoptimized unless -fno-strict-aliasing is used

Perhaps this is the cause of the occasional freezes that you are seeing? The fix would be to change line 308 of ui.c to

extern void ili9341_line(int, int, int, int, uint16_t);

I haven't tested this change yet to see if the freezes disappear.

damib, 2020-01-21:
Hi

Yes, I've noticed the same warning and suspected it could be a possible 
cause of the problem.

I'll do some testing to see it fix the problem.

Damiano

edy555, 2020-01-21:
@andyg24 thanks for the suggestion.
I tested the fix you pointed out. 
Unfortunately, occasional freeze still happens after the fix.