ttrftech/NanoVNA #110: data method returns unexpected data (extra "0.

InventoryTech, 2020-01-12:
I have (infrequently) been getting exceptions when calling data() method. I added some debug logging:

`    def data(self, array = 0):
        self.send_command("data %d\r" % array)
        data = self.fetch_data()
        x = []
        for line in data.split('\n'):
            if line:
                try:
                    d = line.strip().split(' ')
                    x.append(float(d[0])+float(d[1])*1.j)
                except Exception as e:
                    print(e)
                    print("line was " + line)
        return np.array(x)`

And when the problem occurs:

> could not convert string to float: '0.0.1359530'
> line was 0.108219824 0.0.1359530

InventoryTech, 2020-01-13:
Close as duplicate of #111

closed 2020-01-13 by InventoryTech

erinluihing, 2023-09-14:
@InventoryTech were you able to resolve this?