ttrftech/NanoVNA #122: I2C Bus speed

DiSlord, 2020-03-02:
Tru undestand how config I2C bus
`static const I2CConfig i2ccfg = {
  0x00300506, //voodoo magic 400kHz @ HSI 8MHz
  0,
  0
};`
Refer manual for I2C_TIMINGR register not get this value for 400kHz @ HSI 8MHz

If get example values from manual for 500kHz @ HSI 8MHz and change STM32_TIMINGR_SCLL from 6 to 5-3, possible get 7% sweep speed
```
static const I2CConfig i2ccfg = {
  // TIMINGR register initialization. (use I2C timing configuration tool for STM32F3xx and STM32F0xx microcontrollers (AN4235))
  // 500kHz @ HSI 8MHz (Use 26.4.10 I2C_TIMINGR register configuration examples from STM32 RM0091 Reference manual)
  STM32_TIMINGR_PRESC(0U)  |
  STM32_TIMINGR_SCLDEL(1U) | STM32_TIMINGR_SDADEL(0U) |
  STM32_TIMINGR_SCLH(3U)   | STM32_TIMINGR_SCLL(5U),
  // Old values voodoo magic 400kHz @ HSI 8MHz
//  STM32_TIMINGR_PRESC(0U)  |
//  STM32_TIMINGR_SCLDEL(3U) | STM32_TIMINGR_SDADEL(0U) |
//  STM32_TIMINGR_SCLH(5U)   | STM32_TIMINGR_SCLL(6U),
  0,          // CR1 register initialization.
  0           // CR2 register initialization.
};
```
Try test it

DiSlord, 2020-03-02:
Also if change I2C bus clock to 48MHz, 
`#define STM32_I2C1SW                        STM32_I2C1SW_SYSCLK`
And set 400-500kHz timing settings, allow get less noise (good see on 900-1500MHz)

400kHz @ 48MHz I2C clock
```
  STM32_TIMINGR_PRESC(2U)  |
  STM32_TIMINGR_SCLDEL(2U) | STM32_TIMINGR_SDADEL(0U) |
  STM32_TIMINGR_SCLH(16U)   | STM32_TIMINGR_SCLL(19U),
```

500kHz @ 48MHz I2C clock
```
  STM32_TIMINGR_PRESC(2U)  |
  STM32_TIMINGR_SCLDEL(2U) | STM32_TIMINGR_SDADEL(0U) |
  STM32_TIMINGR_SCLH(3U)   | STM32_TIMINGR_SCLL(24U),
```

DiSlord, 2020-03-02:
Also now allow set #define DELAY_CHANNEL_CHANGE 2 and get good measure result

DiSlord, 2020-03-02:
At last, best result give this settings:
I2C HSI = 8MHz
`#define STM32_I2C1SW                        STM32_I2C1SW_HSI`

```
i2ccfg
// I2C clock bus setting: depend from STM32_I2C1SW in mcuconf.h
// STM32_I2C1SW = STM32_I2C1SW_HSI     (HSI=8MHz)
// STM32_I2C1SW = STM32_I2C1SW_SYSCLK  (SYSCLK = 48MHz)
static const I2CConfig i2ccfg = {
  // TIMINGR register initialization. (use I2C timing configuration tool for STM32F3xx and STM32F0xx microcontrollers (AN4235))
  // 400kHz @ SYSCLK 48MHz (Use 26.4.10 I2C_TIMINGR register configuration examples from STM32 RM0091 Reference manual)
//  STM32_TIMINGR_PRESC(5U)  |
//  STM32_TIMINGR_SCLDEL(3U) | STM32_TIMINGR_SDADEL(3U) |
//  STM32_TIMINGR_SCLH(3U)   | STM32_TIMINGR_SCLL(9U),
// 400kHz @ HSI 8MHz (Use 26.4.10 I2C_TIMINGR register configuration examples from STM32 RM0091 Reference manual)
  STM32_TIMINGR_PRESC(0U)  |
  STM32_TIMINGR_SCLDEL(3U) | STM32_TIMINGR_SDADEL(1U) |
  STM32_TIMINGR_SCLH(3U)   | STM32_TIMINGR_SCLL(9U),
// Old values voodoo magic 400kHz @ HSI 8MHz
//  STM32_TIMINGR_PRESC(0U)  |
//  STM32_TIMINGR_SCLDEL(3U) | STM32_TIMINGR_SDADEL(0U) |
//  STM32_TIMINGR_SCLH(5U)   | STM32_TIMINGR_SCLL(6U),
  0,          // CR1 register initialization.
  0           // CR2 register initialization.
};
```
`#define DELAY_GAIN_CHANGE 2`
`#define DELAY_CHANNEL_CHANGE 2`

si5351.c
`#define DELAY_NORMAL 2
#define DELAY_BANDCHANGE 0
#define DELAY_LOWBAND 2`

Try check it

DiSlord, 2020-03-03:
Additional
```
    // div by 6 mode. both PLL A and B are dedicated for CLK0, CLK1
    si5351_set_frequency_fixeddiv(0, SI5351_PLL_A, ofreq, 6, drive_strength, omul);
    si5351_set_frequency_fixeddiv(1, SI5351_PLL_B, freq, 6, drive_strength, mul);
    si5351_set_frequency_fixedpll(2, SI5351_PLL_B, freq / mul * 6, CLK2_FREQUENCY,
                                  SI5351_R_DIV_1, SI5351_CLK_DRIVE_STRENGTH_2MA, 1);
```
Every freq for band 1 and 2 updade send
Chanel 0 settings (SI5351_PLL_A),
Chanel 1 settings (SI5351_PLL_B), and here broke CH2 clock freq (processor, audio codec clock)
Chanel 2 settings (SI5351_PLL_B), restore correct CH2 clock freq

As minimum need need send CH2 settings before, as maximum at some time apply all or not use it

DiSlord, 2020-03-03:
I2C bus speed not define freq stability, it just define data transfer speed to si5351

Found some problems in si5351.c it contain a lot of hack, need undestand how better solve it
At test i get up to 2x speed. Need just minimise I2C data transfer and do correct control

Pmax65, 2020-03-04:
HI DiSlord,

> Possibly this hack used for fix it on band 1?
> 
>     // Set PLL twice on changing from band 2
>     if (current_band == 2) {
>       si5351_set_frequency_fixeddiv(0, SI5351_PLL_A, ofreq, 6, drive_strength, omul);
>       si5351_set_frequency_fixeddiv(1, SI5351_PLL_B, freq, 6, drive_strength, mul);
>     }
> 
As far I know, that has been thought to avoids PLL unlocks at frequencies around 300MHz. 

About the MCLK "arrhythmia":  I'm working to an HW mod to improve the odd harmonics behavior when the DUT is an high order low-pass filter that lets pass frequencies in the fundamental  carrier range stopping the frequencies in its harmonics range and I discovered that the "arrhythmia" is due to the function si5351_disable_output() used into function si5351_set_frequency_with_offset(...).
In fact, function si5351_disable_output() suspends CLK2 output every times the band changes and re-enables it only until si5351_enable_output() is called at the end of si5351_set_frequency_with_offset(...).

In my HW mod (which I ended yesterday and I'll publish soon), I added an external 8MHz oscillator for the MCLK and used CLK2 to get an harmonics only reference signal that fixes the odd harmonics issue. The mod, in few words, uses CLK1 output to source the DUT below 300MHz, while it uses CLK2 output to source the DUT above 300MHz passing the reference signal through a 5th order high-pass elliptic filter (in this case the Si5351A is used as an RF switch, both its outputs generate the very same frequency, but one at time).

Have a great day.

Massimo IK1IZA

DiSlord, 2020-03-04:
Found why need this hack
```
// Set PLL twice on changing from band 2
if (current_band == 2) {
  si5351_set_frequency_fixeddiv(0, SI5351_PLL_A, ofreq, 6, drive_strength, omul);
  si5351_set_frequency_fixeddiv(1, SI5351_PLL_B, freq, 6, drive_strength, mul);
}
```
si5153 have some delay on change MultiSynth Integer and non Integer Mode.
Need send CLK Control twice vs some delay on band change

I cache PLL, MultiSynth, CLK Control settings and update in only if chaned. And need wait some time if CLK Control changed

DiSlord, 2020-03-05:
Seems i solve all hack for si5351, soon i add Pull request

Sweep freq speedup from 8400 sys tick to 5300 (about 60% speedup)
Need just test stability

DiSlord, 2020-03-06:
Try undestand why FREQ_HARMONICS  > 300 not work (si5351 can generate it)
```
  For FREQ_HARMONICS = 300MHz - band range is:
  1: Band 1 = 1-100MHz Band 2 = 100-150MHz, Band 3 = 150- 300MHz
   In harmonic mode 
  3:                   Band 2 = 300-450MHz, Band 3 =  450- 900MHz
  5:                                        Band 3 =  900-1500MHz
  7:                                        Band 3 = 1500-2100MHz
  9:                                        Band 3 = 2100-2700MHz
```
On frequences bigger 300MHz freq and ofreq need separate band selection
```
   Warning!! for FREQ_HARMONICS > 300 in some cases for ofreq selected incorrect band, need separate band selection for freq and ofreq
   *  Example if FREQ_HARMONICS = 500MHz, freq = 510MHz:
   *  f = Band 3, of = Band 2 - and c`ant generate 

```
I check this on next patch

DiSlord, 2020-03-07:
Create pull request

closed 2020-03-07 by DiSlord