ttrftech/NanoVNA #124: No way to exit from the marker/search/minimum (other than reboot)

yarda, 2020-03-06:
Reproducer:
1. marker/search/minimum

No way to return to the normal mode which allows moving of the marker (by clicking the wheel left/right). I tried marker/search/tracking option, switching the marker on/off, nothing helped. It seems the only way how to return to the "free move" mode is reboot. I guess that it's the same with the marker/search/maximum and maybe other modes as well.

yarda, 2020-03-06:
FW 0.7.0-20200223

DiSlord, 2020-03-07:
For return to normal leveler mode which allows moving of the marker (by clicking the wheel left/right). Ned click at top screen (or possible touch not calibrated)

yarda, 2020-03-07:
Yup, clicking the top screen works (I have to tried multiple times, because I wasn't sure where exactly to click). For me this was really non intuitive. Could you mention this in the documentation? Thanks.