ttrftech/NanoVNA #127: 0.7.0-32-gc40d78d Random jitters specific sweep ranges

d51r3verse, 2020-03-11:
```
Version: 0.7.0-32-gc40d78d
Build Time: Mar 10 2020 - 16:06:33
Kernel: 4.0.0
Compiler: GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
```
sweep range combinations like below
```
sweep span 375000000 [or 292500000]
sweep center 300000000 [or 350000000]
```
[image]
This behavior similar issue like 6.5Mhz, 64Mhz, 300Mhz narrow&little spikes( https://github.com/hugen79/NanoVNA-H/issues/14#issuecomment-555456572 ).
But this is too wide(50Mhz) and too serious(+-30dB)

Not happens this wide&huge spikes on 0.7.0-20200223(7d2708a)

DiSlord, 2020-03-11:
Try increase this delay
```
// Reset PLL need then band changes
static void si5351_reset_pll(uint8_t mask)
{
  // Writing a 1<<5 will reset PLLA, 1<<7 reset PLLB, this is a self clearing bits.
  // !!! Need delay before reset PLL for apply PLL freq changes before
  chThdSleepMicroseconds(400);
  si5351_write(SI5351_REG_177_PLL_RESET, mask | 0x0C);
}
```

DiSlord, 2020-03-11:
I test it if enter delay 1000 
Seems work good
`chThdSleepMicroseconds(1000);`

d51r3verse, 2020-03-11:
Huge jitters suppressed.
[image]
@DiSlord Confirmed ```chThdSleepMicroseconds(5000);```. 👍

DiSlord, 2020-03-11:
Can you compare firmware 0.7 and this fixes?
I hope that despite the fact that sweep speed faster stability has improved.
Also as i see on screenshots exist artifacts on screen, it should be fixed at this PR
https://github.com/ttrftech/NanoVNA/pull/126

d51r3verse, 2020-03-11:
Tested PR126(https://github.com/DiSlord/NanoVNA/commit/51b5cce0167f2955363059bcc7557ba75132d275)
Screen updates faster than 0.4.0(50k~2.147G sweep /w all traces enabled)/Screen artifacts also gone💯 

If I have more time, I wish to find another HW/SW nonlinear measure points.
Thnx your contribution

closed 2020-03-31 by d51r3verse