ttrftech/NanoVNA #131: Incorrect battery readings

SMinin, 2020-03-23:
I have a miniature battery from the phone with a nominal voltage of 3.6V. I fully charge the device and after I turn it on the charge indicator shows about 70%. After half an hour the indicator turns red and then works for another two and a half hours. How can I calibrate the battery indicator? Firmware 0.7.1.

DiSlord, 2020-03-23:
Connect to PC, and correct v_bat measure setting (depend from used diode)
Run putty or another terminal program
Reset: 'vbat_offset 0'
Measure: 'v_bat'
Commpare vs Real multimeter measurenent.
Set correction in mV run: 'vbat_offset 150'

For NanoVNA-H model you can use 150

SMinin, 2020-03-23:
Thanks, it all worked. vbat_offset set to 500.

Fireobirdy, 2020-05-02:
I set vbat_offset to 0 and measured with command vbat. It said 3mV. The measurement with my multimeter said 4090mV. Is there something broken?