ttrftech/NanoVNA #133: How it works？

cq2ap9, 2020-04-20:
I saw your NanoVNA open source project, this is a great project. I purchased the NanoVAN kit from China and learned to use it. I analyzed the circuit diagram and program code, but I still don't understand how it works. I have n’t found a description on the Internet. I ’m very curious about the principle used to make the original complicated RF measurement technology so simple and low cost. Could you tell me if there is any working principle and program structure of NanoVNA circuit Documentation, or can you tell me where I can get this knowledge?
thank you very much!

K9CIY, 2020-04-20:
https://nanovna.com/?page_id=87  
The NanoVNA is opensource. You can bind the structure map at the site above and it has links to more information.

DiSlord, 2020-04-21:
You can unerstand how it work only if read source code and scheme

Basic is, si5351 generate 2 frequency freq and freq + offset
Freq go to CH0 reflect and pass to CH1
After it mix on SA612 vs freq+offsset for get
IF referency 
IF reflect
IF pass

After all 3 data go to audio codec TLV320AIC3204, where use DSP for get complex IF freq data
For correct measure need calibrate it on Open, Short, Load, Isolin, Thru modes and recive calibration coefficients

cq2ap9, 2020-04-23:
Thank you,
I know that the reference signal, the reflected signal and the transmission signal are mixed with the LO signal to obtain an IF signal through three frequency mixers. This IF signal is an ordinary IF signal, not two orthogonal IQ signals, right? I am more puzzled that FFT is usually used to calculate the amplitude and phase information of the signal. But I read the source code, I think it is not the FFT method, but the digital intermediate frequency mode, by multiplying Sin and Cos and then accumulating, thereby obtaining the IQ signal, calculating the amplitude and phase information, I do n’t quite understand Hope someone can answer.

DiSlord, 2020-04-24:
Yes you correct in dsp.c on sin/cos table multiple - it DFT for IF frequency
recive 2 complex IQ vector of ref and measure signal data

measured signal less then ref on 'k' and rotate on 'a' so need recive it

result = measure/ref = k*(cos(x+a)+i*sin(x+a))/(cos(x)+i*b*sin(x)) = k*(cos(a)+i*sin(a))
So at the end calculation you recive amplitude and phase information. in next calc ref signal data not use

cq2ap9, 2020-05-05:
Thanks, I understand it in general.
The calculation in dsp.c seems to be the single-point DFT algorithm of the IF signal. In this way, I understand what is going on. Next, I will study the calibration of NanoVNA.