ttrftech/NanoVNA #135: Export of S1P file yields wrong data?

Herman-RFguy, 2020-05-09:
When exporting an S1P file and calculate a data line of corresponding frequency as the marker value back to the marker data displayed i get wrong data.
When perform the same measurement, test set-up, DUT  and calibration on a ZNC3  ( Rohde & Schwarz) i get a different S1P content.
Calculating this back using the same algorithm it yields correct marker data.
[image]

Attached are both S1P files from nano VNA and ZNC3, a screenshot of the sw version ov the nanoVNA.
I would appreciate any help to solve this issue, if it is not already, but i could not find it. -:)

[znc3.s1p.zip](https://github.com/ttrftech/NanoVNA/files/4603586/znc3.s1p.zip)
[nanovna_result.s1p.zip](https://github.com/ttrftech/NanoVNA/files/4603595/nanovna_result.s1p.zip)
[image]

DiSlord, 2020-05-10:
Possibly you not correct read values from file?
NanoVNA
50000 -1.00041697950416 0.007368446112387565
ZNC3
5.000000000000000E4    -1.000193611347694       6.168578964723261E-3
Values ~equal, but ZNC3 use exp format for store

Or get next random string
NanoVNA
29520500 -0.009519632466691198 0.38243573049738105
ZNC3
2.952462809917355E7    -7.411973193649163E-2    3.916175087556830E-1
Also ~equal

Herman-RFguy, 2020-05-10:
Thanks for your hint.
This is not the issue.
The issue is what is coming after the frequency, the content of the S11 value.
There it goes wrong.
Although the headers of both the ZNC and the nanoVNA specify 
# Hz S RI R 50
this seems for the nano VNA

- or not to be true but is possibly is another format or:

- the format is true but the numeric values are errrornous.
I want that to be cleared so i can proceed my calculations.

closed 2020-05-10 by Herman-RFguy