ttrftech/NanoVNA #136: exporting S1P file with sw Nano VNA-Saver yields wrong data

Herman-RFguy, 2020-05-10:
I closed my issue by making a mistake.
Here it is again.
I compared 3 S1P files.
All generated with the same way of calibrating, and the same DUT.
one generated with a R&S ZNC-3, one with nano-VNA and -saver, and one with nanovna and sharp SW.

Calculating from the S1P files back to the marker value readings the R&S ZNC3 and the combination nanoVNA with Sharp succeeds, the combination nanovna with Saver fails.

conclusion: The nanovna-Saver software exports erroneous S1P files.
I enclose all 3 S1P files for investigation.
[sharp.s1p.zip](https://github.com/ttrftech/NanoVNA/files/4605835/sharp.s1p.zip)
[znc3.s1p.zip](https://github.com/ttrftech/NanoVNA/files/4605836/znc3.s1p.zip)
[nanovna_result.s1p.zip](https://github.com/ttrftech/NanoVNA/files/4605837/nanovna_result.s1p.zip)

For a nice application, measuring the complex permeability of toroids by measuring  S11 
it is important to have good data, so my request is a bug fix.
Used SW version info:
[image]

DiSlord, 2020-05-10:
NanoVNA as hardware not export s1p files
Export NanoVNASaver software, it should made correct format of file
You can try ask on NanoVNASaver repo
https://github.com/mihtjel/nanovna-saver
But as i know dewelopwer not ask ¬3 mounth
Or you can ask on NanoVNA forum
https://groups.io/g/nanovna-users

Herman-RFguy, 2020-05-10:
Thank you for your assistance, i will try the given links to find support.
With kind regards,
Herman.

closed 2020-07-23 by Herman-RFguy