ttrftech/NanoVNA #150: NanoVNA not recognized by Python script

0xCoto, 2021-09-26:
@edy555 @cho45 I'm trying to run the Python script, but I get
```
Traceback (most recent call last):
  File "/Users/user/Documents/NetworkOptimizer/./nanovna.py", line 377, in <module>
    nv = NanoVNA(opt.device or getport())
  File "/Users/user/Documents/NetworkOptimizer/./nanovna.py", line 18, in getport
    raise OSError("device not found")
OSError: device not found
```

Even though my NanoVNA does get recognized fine by NanoVNA-Saver, listing: `/dev/cu.usbmodemDEMO1 (S-A-A-2)`

[image]

Anyone faced this issue or has any idea what the issue could be? I also tried debugging using
`print(device_list)`, and I see a difference:

NanoVNA plugged in:
```
[<serial.tools.list_ports_common.ListPortInfo object at 0x10d484be0>, <serial.tools.list_ports_common.ListPortInfo object at 0x10d484ca0>]
```
vs.
Nano VNA not plugged in:
```
[<serial.tools.list_ports_common.ListPortInfo object at 0x119392b80>]
```

Thanks in advance!

0xCoto, 2021-09-27:
Update: I figured from the NanoVNA-Saver that it uses
```python
VID = 0x04b4
PID = 0x0008
```
for the S-A-A V2 model, and although at first it seems to work, it doesn't respond. E.g. I run `./nanovna.py -C out.png` and it gets stuck on `capturing...` without any `out.png` being saved of course. Perhaps there's a better Python script suited S-A-A V2 model?

0xCoto, 2021-12-18:
@zarath One more thing: I've managed to get everything working and I'm able to output touchstone files, plot them etc., but I see that the |S21| values disagree between the NanoVNA alone (i.e. not connected to python/a computer) and the ones the python script is measuring.

My understanding is that once I calibrate the device (SOLT) and connect it to the computer, the python interface does not seem to pick up the latest calibration on file. Is there a simple way to either load the latest calibration on file or manually re-run the SOLT calibration again through python, or do I write something up and save each state (OPEN, SHORT, LOAD, THRU) as a separate `sNp` file and apply the calibration locally?

zarath, 2022-01-04:
Oh, that's of course a lot more work... V2 just delivers uncalibrated data via usb and it is requiered to do it by ones own.
An idea would be to give the script an option to read in a calibration file, which for example can be created by nanoVNA-saver

0xCoto, 2022-01-04:
Solved with scikit-rf's calibration tool, thanks!

Haydarjammoul, 2022-04-26:
Hello
can you share with us how you were able to solve it.
Currently blocked on trying to connect my SAA-2N via python
Thank you in advance

0xCoto, 2022-04-30:
@Haydarjammoul I had to change my VIDPIDs to:
`VIDPIDs = set([(0x0483, 0x5740), (0x04b4,0x0008)]);`
so the SAA V2 device could be recognized. See: https://github.com/0xCoto/OptimizeRF/blob/main/nanovna.py